locusrobotics/fuse #29: Documentation and performance metrics

YoshuaNava, 2018-10-24:
Hi developers,
This sensor fusion approach looks quite attractive. Thank you for your great work. I read the README and slides, and currently wanted to ask you:

* Are you planning on publishing some other reference or paper?
* Have you measured in some way the computational performance of this optimization approach?
* If you have some metrics on its accuracy. (I know fuse is a generic approach, which makes measuring the accuracy somewhat dependable on the plugins you use, but it would be interesting to have some numbers)
* Do you have some examples for running this package?

Thank you.

Yoshua Nava

svwilliams, 2018-10-24:
Apologies for what is going to be an unsatisfying answer. As you may have gathered from the current state of documentation, this project is still in active development. I am working to get better documentation in place and provide a set of tutorials to help everyone get started...but they are not ready just yet.

To address your questions specifically:
> Are you planning on publishing some other reference or paper?

I am not planning any sort of academic publication at this time. I did propose a ROSCon2018 talk, and plan to propose talks for future ROSCons as well. And I will expand the documentation over time.
> Have you measured in some way the computational performance of this optimization approach?

One answer is that the underlying optimization is handled by Google Ceres. Characterizing the performance of Ceres alone is a daunting task all by itself, as there are many parameters that can be tuned, solver types to select from, etc. There are some examples on the [Ceres site](http://ceres-solver.org/nnls_tutorial.html#curve-fitting) that solve some common benchmark data sets. One thing to note is that Ceres implements a batch, Levenberg–Marquardt optimization. If you are running a live, SLAM-type problem, the problem will grow over time and the solve time of each iteration will increase.

Another performance question is how much overhead `fuse` imposes above and beyond the Ceres nonlinear solver. That I can try to characterize in the future, but have not done so yet. Given the computational complexity of the solver itself, I expect the `fuse` overhead to be negligible. But future benchmarks will tell...

> If you have some metrics on its accuracy.

As you point out, the accuracy of the results compared to ground truth will be a function of the sensor models in use, as well as properties of the numerical solver. On the solver side, Google Ceres does provide good numerical accuracy ([here](https://groups.google.com/forum/#!topic/ceres-solver/UcicgMPgbXw)), at least according to the authors of the package.

> Do you have some examples for running this package?

Not yet. I am working on getting a set of tutorials and example problems released. Expect to see those in the next few months. Also, a colleague is working on implementing the [robot_localization](http://docs.ros.org/melodic/api/robot_localization/html/index.html) package as a set of `fuse` sensor and motion model plugins.

If there are specific benchmarks on open-source datasets you would be interest in, please let me know. At the moment I'm thinking of creating examples for the Sphere2500 and Manhattan3500 benchmark datasets.

YoshuaNava, 2018-11-04:
Hi @svwilliams,
Thank you for your reply.

Is not underwhelming at all. I'm tackling a similar problem, and I think your approach is quite promising.  I'll stay tuned for further updates. 

Another question I wanted to ask you, if it isn't much bother, is why you chose Ceres over gtsam, considering all the powerful tools already built in gtsam.

Probably the aspect that concerns me the most of using non-linear optimization for sensor fusion is the accumulation of nodes. Do you plan to implement some kind of strategy for marginalization in the future?

Thank you.

All the best,
Yoshua

facontidavide, 2019-10-30:
Any update about the "fuse version of the robot_localization package."?