locusrobotics/fuse #58: Missing ROS Doc

beetleskin, 2019-05-23:
As a user, it would be nice to have an overview about nodes, parameters, launch-files, API etc. Either on wiki.ros.org/fuse or docs.ros.org, or both.

svwilliams, 2019-05-23:
Agreed. I am putting documentation together, but it is going slowly. Until then, what documentation is available is on the [github page](https://github.com/locusrobotics/fuse/blob/devel/README.md).

tim-fan, 2023-08-08:
It would be nice to have more documentation around how to configure the provided nodes via yaml. The "getting started" tutorial is nice in providing a specific example, but I'm left wondering what other options there are in terms of  motion models, sensor models, publishers etc, and how to go about configuring them. Also I'm not clear about which use-cases can be handled via config, and when I might need to start writing my own cpp to implement a constraint or model etc. 

I'm thinking in comparison about the nav2 docs, there are so many plugins etc available, and for each one you can generally count on there being corresponding documentation explaining how it works, and providing example configuration for it. It would be nice to have something similar for fuse.

tim-fan, 2023-08-08:
I'd also note that most of the links under "further documentation" on the page https://docs.ros.org/en/api/fuse_doc/html/index.html are broken for me, not sure what's happening there.

svwilliams, 2023-08-09:
> I'd also note that most of the links under "further documentation" on the page

Yes. I noticed the same thing recently. I'm looking into it.