locusrobotics/fuse #59: Add Demo-Launchfiles

beetleskin, 2019-05-23:
To get started, it would be nice to have several Demo launch-files with artificial data, that run several flavors/configs of fuse localization, visualize the data in RViz and maybe some plots.

svwilliams, 2019-05-23:
Also agreed. And I will eventually put those together.

I will also caution you that this project is in active development. and I do not consider it a "1.0" release yet. As such, the API will change without notice. I am mostly happy with the current inter-component API,  but there are some subtle changes planned (like some unique ptrs replacing shared ptrs), along with whatever changes are needed as I find and fix issues.

beetleskin, 2019-05-23:
Thanks for the info. Btw, great package, and thanks for putting it online :)

nicolaje, 2019-08-07:
Hello,

this package looks very promising as a general sensor fusion framework.

Could you add a minimal working sample launchfile / simulation so we can get an entry point as to how to use and extend this package ?

Best,

Jeremy

facontidavide, 2019-10-26:
Hi, 

this library looks amazinh, but without a few example, it is really hard to figure out how to get started.

It would be nice to get executable examples related to the ones in the README file

saimouli, 2020-04-10:
Wondering if there are any examples coming anytime soon :-)

svwilliams, 2020-04-10:
Yeah. I need to spend the time to put some sort of turtlebot sim together. Though I still can't say when that will happen. If anyone is willing to put that sim together, I'll gladly walk them through how to configure the fuse system.

saimouli, 2020-04-10:
@svwilliams that would be great. I had exactly the same idea of configuring the pkg with the Turtebot 3. I am actually working on it now. It would be helpful if you would walk me through the process (either through email or as GitHub discussion).

svwilliams, 2020-04-10:
Sure thing. Send me an email at swilliams@locusrobotics.com and we can work out the details.

vinnnyr, 2021-03-15:
@saimouli Has this guide been yet published? :)

Alex-Beh, 2023-08-03:
@saimouli @svwilliams Any update for the guide? Willing to help and learn this great library.