locusrobotics/fuse #73: Question: Support for landmarks

tim-fan, 2019-06-30:
Hi there, this looks like a great package - I look forward to trying it out!

Just wondering if you plan to support landmark observations at some point in the future?

I'm thinking of landmark support more-or-less the same as what cartographer provides, as outlined in this RFC:
https://github.com/googlecartographer/rfcs/blob/master/text/0011-landmarks.md

svwilliams, 2019-06-30:
There might eventually be support for landmarks; it is something I would like do. But it will be some time before I release anything along those lines.

However, the fuse framework has been intentionally designed to be easily extensible. If you have the need and the expertise, you could make your own landmark sensor models that plug into the fuse optimizers. I'll gladly give you some pointers on what needs to be done if you are interested.

tim-fan, 2019-07-01:
Ok great, thanks for the info. I don't have any immediate need for the functionality at the moment, but will keep that in mind as an option for the future.

closed 2019-07-01 by tim-fan