locusrobotics/fuse #82: Support for ROS2

bradsquicciarini, 2019-08-09:
Hello, I was wondering if/when you all intend to extend this to ROS2?

skyler237, 2021-04-19:
Just curious if there is any update here as to whether you plan on migrating to ROS2 or not and if so, what the anticipated timeline might be

svwilliams, 2021-04-19:
I have aspirations of supporting ROS2, but I do not currently have that work scheduled.

BrettRD, 2021-11-01:
I have some time to throw at this.
In the spirit of avoiding duplicate work, does anyone know of any prior attempts at a port?

The architecture of Fuse seems sane enough for a direct port to ROS2, do any established users have a wish-list of things that should be changed?

svwilliams, 2021-11-04:
I have not made any attempt to port this to ROS2 yet, nor have I heard of anyone attempting it before. And sadly, my ROS2 background is very limited at the moment.

Fuse has several components (sensor models, motion models, publishers) that mimic node operations -- sensors "publish" data to an optimizer, a motion model provides a "service" to link requested timestamps together, and a publisher "subscribes" to graph updates. In ROS1 these were implemented as plugins to allow derived class objects (e.g. custom sensor constraints and variable types) to be passed around as base class pointers. I have no idea if ROS2 has a better way doing this, or if plugins are still the best choice. Also, to support these node-like objects in plugin form, I got creative and had the plugins insert function calls into the ROS1 callback queues. I don't know how viable that approach is in ROS2. But otherwise, I don't recall anything too strange, nor are there any big wishlist items I know about.

BrettRD, 2021-11-06:
Pluginlib still seems like a good choice for loading derived classes for constraints and variables into the optimiser.
Because constraints get passed to the optimiser from the *_model classes, the *_models should also be loaded into the optimiser directly (with pluginlib) to avoid serialisation of constraints into and out of a ROS message.

ComposableNodes is a desirable feature for interfacing with high-bandwidth sensors, but I think that'll boil down to a few extra CMake lines around the optimiser.

The callback queue trick has become quite difficult.  The particular calls injected into the callback queue look like they'd be well-served by the ROS2 LifecycleNode system, I'll have a look at how that might hang together, I suspect it'll want a per-model node name to associate the lifecycle services to.

methylDragon, 2023-03-09:
Done!:
- https://github.com/locusrobotics/fuse/issues/276

DLu, 2023-07-24:
Okay to close this issue @svwilliams ?

svwilliams, 2023-07-24:
Yes

closed 2023-07-24 by svwilliams