locusrobotics/fuse #174: catkin_make failure fuse_optimizers

saimouli, 2020-04-10:
```
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp: In member function ‘void fuse_optimizers::Optimizer::loadMotionModels()’:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:166:85: error: ambiguous overload for ‘operator[]’ (operand types are ‘const XmlRpc::XmlRpcValue’ and ‘const char [5]’)
                                         static_cast<std::string>(motion_model_config["type"]), motion_model_config);
                                                                                     ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:97:24: note: candidate: const XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](int) const <near match>
     XmlRpcValue const& operator[](int i) const { assertArray(i+1); return _value.asArray->at(i); }
                        ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:97:24: note:   conversion of argument 1 would be ill-formed:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:166:92: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
                                         static_cast<std::string>(motion_model_config["type"]), motion_model_config);
                                                                                            ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:98:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](int) <near match>
     XmlRpcValue& operator[](int i)             { assertArray(i+1); return _value.asArray->at(i); }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:98:18: note:   conversion of argument 1 would be ill-formed:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:166:92: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
                                         static_cast<std::string>(motion_model_config["type"]), motion_model_config);
                                                                                            ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:100:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const string&) <near match>
     XmlRpcValue& operator[](std::string const& k) { assertStruct(); return (*_value.asStruct)[k]; }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:100:18: note:   passing ‘const XmlRpc::XmlRpcValue*’ as ‘this’ argument discards qualifiers
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:101:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const char*) <near match>
     XmlRpcValue& operator[](const char* k) { assertStruct(); std::string s(k); return (*_value.asStruct)[s]; }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:101:18: note:   passing ‘const XmlRpc::XmlRpcValue*’ as ‘this’ argument discards qualifiers
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp: In member function ‘void fuse_optimizers::Optimizer::loadSensorModels()’:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:214:78: error: ambiguous overload for ‘operator[]’ (operand types are ‘const XmlRpc::XmlRpcValue’ and ‘const char [5]’)
       sensor_model_configs.emplace_back(static_cast<std::string>(sensor_model["name"]),
                                                                              ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:97:24: note: candidate: const XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](int) const <near match>
     XmlRpcValue const& operator[](int i) const { assertArray(i+1); return _value.asArray->at(i); }
                        ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:97:24: note:   conversion of argument 1 would be ill-formed:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:214:85: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
       sensor_model_configs.emplace_back(static_cast<std::string>(sensor_model["name"]),
                                                                                     ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:98:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](int) <near match>
     XmlRpcValue& operator[](int i)             { assertArray(i+1); return _value.asArray->at(i); }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:98:18: note:   conversion of argument 1 would be ill-formed:
/mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:214:85: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
       sensor_model_configs.emplace_back(static_cast<std::string>(sensor_model["name"]),
                                                                                     ^
In file included from /opt/abunt/include/ros/node_handle.h:52:0,
                 from /opt/abunt/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/include/fuse_optimizers/optimizer.h:37,
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:100:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const string&) <near match>
[... 295 lines cut ...]
                 from /mnt/943ec2c2-6cda-4cd0-887c-1fed9ec45b2e/fuse_ws/src/fuse/fuse_optimizers/src/optimizer.cpp:38:
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:100:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const string&) <near match>
     XmlRpcValue& operator[](std::string const& k) { assertStruct(); return (*_value.asStruct)[k]; }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:100:18: note:   passing ‘const XmlRpc::XmlRpcValue*’ as ‘this’ argument discards qualifiers
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:101:18: note: candidate: XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const char*) <near match>
     XmlRpcValue& operator[](const char* k) { assertStruct(); std::string s(k); return (*_value.asStruct)[s]; }
                  ^~~~~~~~
/opt/abunt/include/xmlrpcpp/XmlRpcValue.h:101:18: note:   passing ‘const XmlRpc::XmlRpcValue*’ as ‘this’ argument discards qualifiers
[... cut ...]
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```
XmlRpcValue has the following overload operators:
```
XmlRpcValue const& operator[](int i) const { assertArray(i+1); return _value.asArray->at(i); }
    XmlRpcValue& operator[](int i)             { assertArray(i+1); return _value.asArray->at(i); }

    XmlRpcValue& operator[](std::string const& k) { assertStruct(); return (*_value.asStruct)[k]; }
    XmlRpcValue& operator[](const char* k) { assertStruct(); std::string s(k); return (*_value.asStruct)[s]; }
```

svwilliams, 2020-04-10:
What OS and ROS version are you using?

saimouli, 2020-04-10:
OS: 5.3.0-46-generic #38~18.04.1-Ubuntu x86_64 x86_64 x86_64 GNU/Linux

ROS: melodic

svwilliams, 2020-04-10:
Hmm. That is very similar to my machine. Give me a bit to stand up a clean VM and see what happens.

svwilliams, 2020-04-10:
I made a clean Ubuntu 18.04 + ROS melodic + fuse VM, and everything compiles just fine. In the above error I note that you are using a non-standard version of XMLRPC:
`/opt/abunt/include/xmlrpcpp/XmlRpcValue.h`
Can you verify what version that is? Or reconfigure your workspace to use the version included with ROS?

saimouli, 2020-04-10:
version: release/melodic/xmlrpcpp/1.14.2-0

what version do you recommend?

saimouli, 2020-04-10:
xmlrpcpp version 1.14.5-1 worked for me

closed 2020-04-10 by saimouli