locusrobotics/fuse #197: ROS[Kinetic - ubuntu 16.04] Build failure : Ceres package not found.

sarath-vanga, 2020-06-01:
Hi,
I am trying to compile the 'fuse' library on Ubuntu 16.04 with ROS Kinectic installed, but it fails saying the 'Ceres' package is not found. Can someone help me with the 'Ceres-solver' ROS package ?

svwilliams, 2020-06-01:
To the best of my knowledge, Ceres-Solver is available in 16.04.
* I see an Ubuntu 16.04 package here: https://packages.ubuntu.com/xenial/libdevel/libceres-dev
* I see a rosdep entry for it here: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml
* And just to make sure, I double checked that I'm specifying the correct rosdep name here:
https://github.com/locusrobotics/fuse/blob/devel/fuse_core/package.xml#L15

But admittedly I'm running Bionic, so I do not test the Xenial version as often as I should.
* Can you check if you have the Ubuntu package `libceres-dev` installed?
* Also, can you verify if running rosdep to install the package dependencies succeeds? If not, can you send me the error? Something like: `$ rosdep install --from-paths ~/catkin_workspace/src/fuse --ignore-src -r -y`

sarath-vanga, 2020-06-01:
Thanks for the quick response.

The Ubuntu libceres-dev package is not installed on my system, but using "rosdep install" as you suggested worked and I was able to proceed with the compilation. 

Now I encounter the following compiler error:
 In member function ‘virtual bool fuse_core::DelayedThrottleFilter::isEnabled()’:
/home/test/comm_ws/src/fuse-devel/fuse_core/include/fuse_core/console.h:78:58: error: ‘ROSCONSOLE_THROTTLE_CHECK’ was not declared in this scope
     if (ROSCONSOLE_THROTTLE_CHECK(now, last_hit_, period_))

svwilliams, 2020-06-01:
Hmmm. I see `ROSCONSOLE_THROTTLE_CHECK` is included in ROS melodic:
https://github.com/ros/rosconsole/blob/melodic-devel/include/ros/console.h#L369

What version of ROS are you using? I'm guessing Kinetic?

svwilliams, 2020-06-02:
I setup a clean Xenial/Kinetic VM to test, and I did reproduced your compile error. I also found a few XMLRPC errors you'd encounter after that. I created a branch called `kinetic-devel` with the required changes. If you are still interested, give that branch a try and let me know if that works for you as well. If so, I'll need to decide how exactly how I want to support Xenial/Kinetic moving forward. But hopefully this will get you running in the mean time.

sarath-vanga, 2020-06-04:
The new branch (kinetic-devel) worked for me. Thanks for the support.

closed 2020-06-09 by sarath-vanga