locusrobotics/fuse #223: Cannot create constraints for IMU Twist messages with less than 2ms delay

meghdeepj, 2021-02-06:
Board: Nvidia Jetson Nano (Jetpack 4.4)
OS: Ubuntu 18.04
ROS: Melodic

Fuse sounds very promising and I am eager to use it for my AMR navigation stack. I am using fuse to combine data from Wheel odometry and IMU readings. IMU data is coming from Intel D435i via Madgwick Filter.
My launch file and `smoother.yaml` file is as follows:
 
 ```
<node name="smoother" pkg="fuse_optimizers" type="fixed_lag_smoother_node" output="screen">
        <rosparam command="load" file="$(find loc_mod)/params/fuse_imu.yaml" subst_value="true"/>
```
```
#Fixed-lag smoother configuration
optimization_frequency: 10  # The best-effort frequency (in Hz) of the optimization loop. This means that the optimizer will run every 0.1s and trigger all of the publishers when it is done
lag_duration: 0.5  # The size of the fixed-lag window, in seconds. The larger the window, the more variables will be inside the optimizer at any given time, which affects CPU usage.

# Define the set of motion model plugins to load
# Currently fuse only includes a 2D motion model. This is roughly equivalent to the robot_localization state, including the linear and angular position, the linear and angular velocity, and the linear acceleration. A turtlebot is a 2D robot, so this is fine.
# Multiple motion models can be loaded, and each sensor defines which set of motion models are used. We only have one motion model currently, and all sensors should use it.
motion_models:
  unicycle_motion_model:
    type: fuse_models::Unicycle2D

# Next define the set of sensor plugins to load
# You naturally want an IMU sensor and an Odometry sensor
# Additionally, you may want an "ignition" sensor. Ignition sensors tell the optimizer when to start running and what the initial state should be. fuse ships with one ignition sensor that initializes the Unicycle2D state variables.
sensor_models:
  ignition_sensor:
    type: fuse_models::Unicycle2DIgnition
    motion_models: [unicycle_motion_model]
    ignition: true
  odometry_sensor:
    type: fuse_models::Odometry2D
    motion_models: [unicycle_motion_model]
  imu_sensor:
    type: fuse_models::Imu2D
    motion_models: [unicycle_motion_model]

# And finally, define the set of publishers. fuse_models includes a robot_localization like publisher that publishes the latest state to the odom and tf topics.
publishers:
  odom_publisher:
    type: fuse_models::Odometry2DPublisher

# That is it for the optimizer itself, but each plugin has its own set of configuration. That configuration is specified in a sub-namespace of the plugin's name. E.g. ignition_sensor, odometry_sensor, imu_sensor
# These parameters are not strictly backwards-compatible with robot_localization configuration, but basically all of the same functionality exists.
unicycle_motion_model:
  buffer_length: 1.0  # How long to keep a local history. This should be at least as long as the smoother lag
  process_noise_diagonal: [0.001, 0.001, 0.001, 0.2, 0.2, 0.2, 0.15, 0.001]  # These are in order {x, y, yaw, vx, vy, vyaw, ax, ay}. And note these are variances. The numerical values should be adjusted to the platform.

ignition_sensor:
  publish_on_startup: true  # Just start immediately. If false, it waits for a service to send it initial state values
  initial_state: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]  # The same order as the process noise above
  initial_sigma: [0.000001, 0.000001, 0.000001, 10.0, 10.0, 10.0, 1.0, 0.000001]   # The same order as the process noise above. Note these are standard deviations. The numerical values should be adjusted to the platform.

odometry_sensor:
  topic: /odom/wheel
  twist_target_frame: base_link
  linear_velocity_dimensions: ['x', 'y']
  angular_velocity_dimensions: ['yaw']
  # The above configuration only uses the odometry velocity information. There are several options for using the odometry pose information as well.

imu_sensor:
  topic: /imu/data
  angular_velocity_dimensions: ['yaw']
  linear_acceleration_dimensions: ['x']
  twist_target_frame: base_link
  # The above configuration only uses the yaw velocity of the IMU measurement. Depending on the IMU data available, additional measurement dimensions can be added

odom_publisher:
  topic: odom
  base_link_frame_id: base_link
  odom_frame_id: odom
  map_frame_id: map
  world_frame_id: odom
  publish_tf: true
  # The odometry publisher also has several configuration modes, again very similar to the robot_localization package. The exact settings will depend on whether you are trying to fuse multiple dead reckoning sensors to make a better "odom" source, or if you are fusing in global state data and want to publish the map->odom transform.

```
The delay in the IMU time stamp is hardly +-0.001~0.002 ms and I am getting the following error on running the lag_smoother.

```
[ERROR] [1612442903.265861936]: Cannot create constraint from twist message with stamp 1612442903.265639067
[ERROR] [1612442903.269917145]: Cannot create constraint from twist message with stamp 1612442903.268141985
[ERROR] [1612442903.270205582]: Cannot create constraint from twist message with stamp 1612442903.270646095
[ERROR] [1612442903.273827301]: Cannot create constraint from twist message with stamp 1612442903.273149014
[ERROR] [1612442903.277885374]: Cannot create constraint from twist message with stamp 1612442903.275653124
[ERROR] [1612442903.278127666]: Cannot create constraint from twist message with stamp 1612442903.278156042
[ERROR] [1612442903.282078759]: Cannot create constraint from twist message with stamp 1612442903.280660152
[ERROR] [1612442903.282325686]: Cannot create constraint from twist message with stamp 1612442903.283163071
[ WARN] [1612442903.282408082]: Optimization exceeded the configured duration by 0.003019425s
[ERROR] [1612442903.285876728]: Cannot create constraint from twist message with stamp 1612442903.285665989
[ERROR] [1612442903.289928499]: Cannot create constraint from twist message with stamp 1612442903.288170099
[ERROR] [1612442903.290246988]: Cannot create constraint from twist message with stamp 1612442903.290674210
```
Thanks for your help.

svwilliams, 2021-02-07:
First off, thanks for your interest in the fuse project, and for fighting your way through the lack of documentation. One day I will finish writing the docs and put together real tutorials.

Now, on to your question:
It looks like the error message is originating here:
https://github.com/locusrobotics/fuse/blob/devel/fuse_models/include/fuse_models/common/sensor_proc.h#L937
...which would arise if the the `transfromMessage()` failed to transform the IMU data into the base_link frame.
https://github.com/locusrobotics/fuse/blob/devel/fuse_models/include/fuse_models/common/sensor_proc.h#L223-L239

Additionally, I would expect to see a warning message every 5s or so that says something to the effect of "Could not transform message from imu_link to base_link. The error was: XXX".

Is the base_link->imu_link frame a static transform?

For a planar robot, I would generally expect the base_link->imu_link frame to be a static transform. If so, then that transform should always be valid, and the tf lookup should always succeed. Are you publishing the transforms in that chain using a tf2 static transform publisher (which uses the `/tf_static` topic), and not a tf1 static transform publisher (which publishes the same transform over and over again at a high rate)?

But if the IMU is mounted on an arm or other articulated structured, then a full tf lookup may be needed. If so, then the problem is the IMU is publishing faster/earlier than the required tf frame broadcaster. Currently the fuse code does not use a tf timeout duration in the `lookupTransform()` call. I should add that as an option, and will try to post a test branch later today. However, that will potentially cause the IMU processing rate to drop. If the IMU sensor has to wait several milliseconds for a tf frame transform to become available, then you will likely drop IMU measurements while you are waiting. Depending on your system, that may or may not be an issue for you.

svwilliams, 2021-02-07:
I created a test branch (`issue223-tf-timeout`) that adds a "tf_timeout" property to the sensor configuration. If this is non-zero, then the tf lookups will wait the configured amount of time before failing the lookup. This _may_ fix your problem, but it may also introduce other problems due to the added processing delay.
https://github.com/locusrobotics/fuse/tree/issue223-tf-timeout

If you try it out, let me know if it fixes your problem. I don't have a dataset with non-static transforms as part of my test suite yet, so I have not verified this will indeed fix your issue.

meghdeepj, 2021-02-08:
Thanks svwilliams for your response.

I am using it on a mobile robot so my imu->base_link is a static transform which is being published by the Intel realsense-ros package. I have even tried with publishing the tf with imu_madgwick_filter but the error remains the same.

One interesting observation is that when I'm sending only the 'x linear acceleration' ( and not 'yaw angular velocity' twist) dimension to the fixed lag smoother from the IMU, the "Cannot create constraint" error doesn't show but the warning: [ WARN] [1612442903.282408082]: Optimization exceeded the configured duration by 0.003019425s still shows up but once in every 10 seconds.

What I could infer is that the issue is mostly due to some messages from the IMU having timestamps with delay -1ms, causing 'Extrapolate to future' error on lookupTransform(). 

I will try out the "tf-timeout test branch" and let you know. I will also try to dig around the source code "fuse_models/common/sensor_proc.h".

Thanks

svwilliams, 2021-02-08:
Okay. I'll take a look at how the Realsense publishes tf frames and IMU data. But if the Realsense is publishing data from the future, I'm less clear what the correct solution should be. I'll let you know if I find anything.

As for the "Optimization exceeded the configured duration by 0.003019425s", that is an indication that the optimization loop exceeded your 10Hz (100ms) time budget. Aside for the obvious of reducing the lag duration or decreasing the sensor input rate, you can also try limiting the number optimizer iterations.

```
...
lag_duration: 0.5
solver_options:
  max_num_iterations: 5
...
```

And I have a long-running goal of optimizing both the CPU time and latency of the system. So if you find any tricks to reduce the optimization time, let me know.

meghdeepj, 2021-02-08:
Test branch `issue223-tf-timeout` that adds a "tf_timeout" has solved the issue.
Keeping the timeout at 100-200ms completely eliminates the "cannot create constraints ERROR" error, but would affect the processing.
Keeping it at ~30-50ms makes the ERROR popup hardly once in every 8-10 seconds, but seems to be a good middle ground for the processing.
Thanks for your help

closed 2021-02-10 by meghdeepj