locusrobotics/fuse #225: Odometry sensor namespace error

meghdeepj, 2021-02-16:
I am using the "issue223-tf-timeout" branch that added the `tf_timeout` parameter to the sensor model, it is working well. However, I came across an error while trying to add a second odometry source by the name `odometry_sensor1`. But, the fixed_lag_smoother is expecting a mistyped namespace `odometert_sensor1`. Changing the namespace to that in the smoother.yaml solved the problem.

The devs might want to look at the expected namespace for multiple odometry sources.

svwilliams, 2021-02-16:
Can you post your configuration file? fuse does not dictate the namespace names. Instead, it uses the user-defined sensor names in the config. My quick attempt to reproduce your error has been unsuccessful. Perhaps loading your exact config will get me somewhere.

meghdeepj, 2021-02-18:
sure, here is the config file
```
# Fixed-lag smoother configuration
optimization_frequency: 10  # The best-effort frequency (in Hz) of the optimization loop. This means that the optimizer will run every 0.1s and trigger all of the publishers when it is done
lag_duration: 0.5  # The size of the fixed-lag window, in seconds. The larger the window, the more variables will be inside the optimizer at any given time, which affects CPU usage.
solver_options:
  max_num_iterations: 5

# Define the set of motion model plugins to load
# Currently fuse only includes a 2D motion model. This is roughly equivalent to the robot_localization state, including the linear and angular position, the linear and angular velocity, and the linear acceleration. A turtlebot is a 2D robot, so this is fine.
# Multiple motion models can be loaded, and each sensor defines which set of motion models are used. We only have one motion model currently, and all sensors should use it.
motion_models:
  unicycle_motion_model:
    type: fuse_models::Unicycle2D

# Next define the set of sensor plugins to load
# You naturally want an IMU sensor and an Odometry sensor
# Additionally, you may want an "ignition" sensor. Ignition sensors tell the optimizer when to start running and what the initial state should be. fuse ships with one ignition sensor that initializes the Unicycle2D state variables.
sensor_models:
  ignition_sensor:
    type: fuse_models::Unicycle2DIgnition
    motion_models: [unicycle_motion_model]
    ignition: true
  odometry_sensor0:
    type: fuse_models::Odometry2D
    motion_models: [unicycle_motion_model]
  odometert_sensor1:
    type: fuse_models::Odometry2D
    motion_models: [unicycle_motion_model]

# And finally, define the set of publishers. fuse_models includes a robot_localization like publisher that publishes the latest state to the odom and tf topics.
publishers:
  odom_publisher:
    type: fuse_models::Odometry2DPublisher

# That is it for the optimizer itself, but each plugin has its own set of configuration. That configuration is specified in a sub-namespace of the plugin's name. E.g. ignition_sensor, odometry_sensor, imu_sensor
# These parameters are not strictly backwards-compatible with robot_localization configuration, but basically all of the same functionality exists.
unicycle_motion_model:
  buffer_length: 1.0  # How long to keep a local history. This should be at least as long as the smoother lag
  process_noise_diagonal: [0.001, 0.001, 0.001, 0.2, 0.3, 0.3, 0.001, 0.001]  # These are in order {x, y, yaw, vx, vy, vyaw, ax, ay}. And note these are variances. The numerical values should be adjusted to the platform.

ignition_sensor:
  publish_on_startup: true  # Just start immediately. If false, it waits for a service to send it initial state values
  initial_state: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]  # The same order as the process noise above
  initial_sigma: [0.000001, 0.000001, 0.000001, 10.0, 10.0, 10.0, 1.0, 0.000001]  # The same order as the process noise above. Note these are standard deviations. The numerical values should be adjusted to the platform.

odometry_sensor0:
  topic: /odom/wheel_fuse
  twist_target_frame: base_link
  linear_velocity_dimensions: ['x', 'y']
  angular_velocity_dimensions: ['yaw']
  # The above configuration only uses the odometry velocity information. There are several options for using the odometry pose information as well.

odometry_sensor1:
  topic: /rs_t265/odom/sample
  twist_target_frame: base_link
  tf_timeout: 0.05 
  linear_velocity_dimensions: ['x']
  angular_velocity_dimensions: ['yaw']

odom_publisher:
  topic: odom
  base_link_frame_id: base_link
  odom_frame_id: odom
  map_frame_id: map
  world_frame_id: odom
  publish_tf: true
  # The odometry publisher also has several configuration modes, again very similar to the robot_localization package. The exact settings will depend on whether you are trying to fuse multiple dead reckoning sensors to make a better "odom" source, or if you are fusing in global state data and want to publish the map->odom transform.

```

Here, `odometry_sensor1:` shows namespace error. But when I change it to `odometert_sensor1:` it runs fine.

svwilliams, 2021-02-18:
The sensor configuration namespaces are defined by the names given in the `sensor_models` dictionary. This block from your config file:
```
sensor_models:
  ignition_sensor:
    type: fuse_models::Unicycle2DIgnition
    motion_models: [unicycle_motion_model]
    ignition: true
  odometry_sensor0:
    type: fuse_models::Odometry2D
    motion_models: [unicycle_motion_model]
  odometert_sensor1:
    type: fuse_models::Odometry2D
    motion_models: [unicycle_motion_model]
```
So you have created three sensors: "ignition_sensor", "odometry_sensor0", and "odometert_sensor1".
And that last one is where the name typo occurs: "odometert_sensor1" instead of "odometry_sensor1". Note that these names can be any valid string, and are not limited to the pattern "odometry_sensorX". If you fix the typo in the above block, it should search for the configuration parameters in the expected namespace.

meghdeepj, 2021-02-19:
My bad, I completely missed that.

Thanks for you help.

closed 2021-02-19 by meghdeepj