locusrobotics/fuse #232: Holding variables through transactions

jakemclaughlin6, 2021-07-20:
When using a graph directly it is possible to hold a variable constant during optimization. But when using the optimizer and sensor models, the main and only form of communication to the main graph is through transactions. Is it possible to notify the optimizer to hold a specific variable constant through a transaction?

svwilliams, 2021-07-21:
Currently there is not. The issue I ran into trying to support variable holds is the general distributed nature of the fuse sensor models. Each sensor model generates a copy of the involved variables and sends it to the optimizer. It is possible that multiple sensors will create the same variable. Perhaps you have a laserscan and are performing both scan-to-scan matches as well as scan-to-map matches. Since both measurements arise from the same raw data, they will involve the same timestamps and thus the same variables. The optimizer receives both transactions and assembles them into a graph. The optimizer/graph is aware that multiple sources may add the same variable. Currently two instances of the same variable are interchangeable, so it just keeps the first copy.

My initial thought was to add a held/not held property to the variable and allow the sensor models to set that flag when desired. However, a problem would arise if the scan-to-scan sensor marked a variable as "held constant" but the scan-to-map sensor added that same variable as "not held constant". It is not clear to me the proper way to deconflict that situation at the optimizer/graph, and so I have not implemented it yet.

Can you describe your use-case to me? If I better understand what you are trying to accomplish, it may become clear how and where to implement the requested variable hold functionality.

jakemclaughlin6, 2021-07-21:
I am attempting to implement visual slam with some already known and optimized landmark locations from a previous optimization, and in the case where the current image sees one of these previous landmarks then I would like to add it to my new optimization, but keep that landmark constant to keep the maps consistent. My current way around this is to just make another re-projection constraint but store the landmark location within the constraint itself as an unchanging Vector3d and optimize against that, rather than as a fuse variable.

svwilliams, 2021-07-21:
One option that I _think_ would work:
fuse would:
* Add an isHeldConstant() or such method to the Variable base class, with a default value of false.
* Update the optimizers/graphs to check that method and configure the Ceres Problem accordingly
You would:
* Derive a custom landmark variable type, FixedLandmark or such, using the Position3d variable as the base class. That class would override the isHeldConstant() method to return true.

In this way, the isHeldConstant() property is defined at compile-time and no deconfliction strategy should be required at runtime.

It means you must know the set of constant landmarks, and must create constant landmarks using a different variable type than new landmarks. Does that seem reasonable? If so, I should be able to put that together shortly.

And out of curiosity, have you created your own variable types for the landmarks already? A stamped position doesn't really make sense for a landmark. It sort of exists for all time. A version that accepted a database ID of some sort might make more sense.

jakemclaughlin6, 2021-07-21:
Yes I have made a new variable for landmarks that is just Position 3d and it takes in a uint64_t as an id to use for creating the uuid. That solutions seems to make sense to me. It is known prior to creation of a landmark if it is to be held constant or not.

This brings up another question I was actually curious about for the marginalization of variables when using the fixed lag smoother. Since the marginalization is based upon timestamps, will the landmarks be successfully marginalized out as well since they are connected to poses which have stamps?

jakemclaughlin6, 2021-07-21:
I have made a PR for the landmark variable, if you would like to review it

svwilliams, 2021-07-21:
Excellent. Thank you so much. I'll get that reviewed shortly.

svwilliams, 2021-07-26:
Just a status report. I have the easy part of this task implemented -- added a method to the Variables class, update the Graph implementation to use that property, created a "fixed" version of the landmark variable. Work in progress is here: https://github.com/locusrobotics/fuse/tree/RST-4390-variable-hold

However, I am convinced the marginalization code used in the fixed-lag smoother will need to be updated to support constant variables. I just haven't quite worked out the details yet. I'll let you know when that is ready and you can test it out.

svwilliams, 2021-08-11:
I just pushed a potential solution for marginalizing fixed variables. Unfortunately there are still a few quirks. Even though a landmark variable is held constant, you will still need to add a prior constraint on the variable. The actual covariance of that prior doesn't affect the solution at all, since the variable is held constant. However, due to some assumptions that are made in the marginalization code, the jacobian blocks from the prior need to exist. Otherwise some matrix sizes are different than expected. I have an idea on how to refactor the marginalization code to eliminate this issue, but that will be a major refactor.

I've added the required changes to this branch, as well as a unit test checking that it works.
https://github.com/locusrobotics/fuse/tree/RST-4390-variable-hold

Give that a try and let me know if it gives you any issues. I'll create a backport branch into the xenial/kinect version as well, but that will take me another couple of days as I need to setup a Xenial VM before I can test it.

svwilliams, 2021-08-14:
And here is a backport branch for Xenial/Kinetic
https://github.com/locusrobotics/fuse/tree/RST-4390-variable-hold-backport

jakemclaughlin6, 2021-08-24:
@svwilliams will this be merged into the main kinetic branch?

jakemclaughlin6, 2021-08-24:
or could the most recent kinetic branch be pulled into RST-4390-variable-hold-backport?

svwilliams, 2021-08-24:
I will merge this if it is working for you. I only have one really simple unit test that is exercising this code. I was hoping for some feedback that it was functioning as expected in real-world use.

jakemclaughlin6, 2021-08-24:
It seems to work for me as I can optimize the poses while keeping landmarks constant, however I have not been able to fully test the functionality of when said fixed landmarks are to be marginalized out and how that effects the optimization, but as of now it seems to work well

svwilliams, 2021-08-24:
Okay. I'll merge this in shortly. And if the fixed-landmark marginalization causes you problems in the future, just open another issue and we can resolve it then.

jakemclaughlin6, 2021-08-24:
sounds good to me

svwilliams, 2021-08-24:
Merged.

closed 2021-08-24 by svwilliams