locusrobotics/fuse #242: Location of throttled_callback.h inconsistent across branches

adthoms, 2021-08-06:
My team is developing a SLAM package to work with ROS Kinetic, Melodic, and (if we are able to resolve compile issues with our proprietary library) Noetic. Right now we have to develop our SLAM package on two seperate branches, one for Kinetic and one for Melodic, since the location of throttled_callback.h in fuse is different between the kinetic-devel and devel branches. In kinetic-devel, the location of throttled_callback.h is:

```
<fuse_models/common/throttled_callback.h>
```

where as in devel, the location of throttled_callback.h is:

```
<fuse_core/throttled_callback.h>
```

Would it be possible to pull throttled_callback.h from the devel branch into the kinetic-devel branch? This will also allow us to make use of [ThrottledMessageCallback](https://github.com/locusrobotics/fuse/blob/3bbf7cdfd159ccea5e6535a0dc7453a6d40fc163/fuse_core/include/fuse_core/throttled_callback.h#L208) in our SLAM package. Once rectified, we may develop our SLAM package on one branch.

svwilliams, 2021-08-06:
That should be possible, yes. I'll let you know when that's ready.

svwilliams, 2021-08-12:
I've moved the location of throttled_callback.h into the fuse_core package in this branch:
https://github.com/locusrobotics/fuse/tree/move-throttle-callback

But do note that there is a slight difference in the implementation between the kinetic-devel version and the current devel version. Depending on the exact issue (if any), it may be possible to get the two implementations closer without breaking backwards compatibility.

Let me know if this meets your needs, or what new problems arise.

adthoms, 2021-08-12:
Thank you so much! everything looks in order, but I will do some testing on my end and I will let you know if other problems arise. I should have this completed by this coming monday

adthoms, 2021-08-17:
This definitely resolves the location of throttled_callback.h between the kinetic-devel and devel branches, though for our SLAM package we still need to maintain separate branches with this change. In a Virtual Environment with a clean install of Ubuntu 16.04, ROS Kinetic, and all other dependencies required by fuse, I was able to compile fuse (using `catkin build` in my catkin workspace) and successfully compile and run all tests in fuse_core (using `catkin build --make-args tests -- fuse_core`) on the move-throttle-callback branch by:

1. Substituting `throttled_callback.h` with `throttled_callback.h` from the devel branch
2. Changing instances of `... = fuse_core::ThrottledCallback<ros_msg::Type>` to `... = fuse_core::ThrottledMessageCallback<ros_msg::Type>` in: acceleration_2d.h, imu_2d.h, odometry_2d.h, pose_2d.h, twist_2d.h,  and test_throttled_callback.cpp
3. Changing instances of `... = node_handle_.subscribe(...` to `... = node_handle_.subscribe<ros_msg::Type>(...` in acceleration_2d.cpp, imu_2d.cpp, odometry_2d.cpp, pose_2d.cpp, twist_2d.cpp, where `ros_msg::Type` matches ros_msg::Type specified in `fuse_core::ThrottledMessageCallback<ros_msg::Type>`

With these changes, we are able to instantiate the templated ThrottledMessageCallback class as well as node handle subscribers consistently across fuse’s devel and kinetic-devel branches, allowing us to develop our SLAM package on one main branch. I can make a pull requested with these changes, though I am not sure where to look for formatting guidelines.

svwilliams, 2021-08-17:
The issue with that approach is that it breaks any `node_handle.subscribe()` calls that may exist out in the world, and I'd rather not do that in an old release. Let me see if I can convince the compiler to auto-deduce the message type when using something closer to the noetic/devel version of the throttled callback.

adthoms, 2021-08-17:
I see, our team has that kind of flexibility but you are right it can break other packages that are dependent on the current kinetic-devel branch. In the meantime I will try to convince my team members to upgrade to Ubuntu 18.04, but they are all too familiar with the difficulties that come with upgrading haha. Thanks again for your help :)

svwilliams, 2021-08-19:
Try https://github.com/locusrobotics/fuse/tree/move-throttle-callback-take-2

I pulled in the devel version of throttled_callback.h into fuse_core, and updated all of the models to use the fuse_core version. But I left the previous throttled_callback.h in place in fuse_models, with a compile-time note that it is deprecated. This should maintain backwards compatibility with any external code.

adthoms, 2021-08-20:
This works wonderfully, all good to merge into kinetic-devel. Thank you!

svwilliams, 2021-08-24:
Merged