locusrobotics/fuse #265: Infinite wait for onStop() to be called

jakemclaughlin6, 2022-01-27:
I am using the reset service server to reset my slam system when I detect a failure, this is so I can automatically reset my initialization sensor models and odometry sensor model simultaneously. While my initialization sensor model stops just fine, my odometry seems to wait infinitely for the result in the stop() function (it is an async sensor model). Is there anything I can do to debug this,or if there is any common causes for this problem?

svwilliams, 2022-01-31:
My first instinct is there is a threading problem -- a callback is blocking waiting for the reset to complete, but since the reset logic executes within the same callback thread, you end up with a deadlock. This same thing happens in a standard single-threaded ROS node if you call a node's service from within a node's message callback.

Where are you calling the reset service from? Are you using the fuse_models odometry sensor, or your own custom version?

svwilliams, 2022-01-31:
It is possible to use configure the base AsyncXXX class to use multiple threads to handle callbacks...but that will have ramifications (mutex guards, etc) on all of the callback methods.

You could also create a callback queue for just the odom messages and spin that with a separate spinner or manual loop. But again, you will have multiple threads running different parts of your sensor model code.

An ugly solution is to make a little loop inside the sensor model callback and call spinOnce() until the reset call completes.

None of these are great solutions.

jakemclaughlin6, 2022-01-31:
Is it possible for there to be an additional reset "service" that is just a subscriber on a topic, and the system is reset when anything publishes on that topic?

svwilliams, 2022-01-31:
It should be easy to add that, yes. Subscribe to an Empty message (or similar) topic and have the message callback execute the same reset logic as the service. Obviously the publisher will (a) have no guarantee the message arrives, and (b) will not know when the reset has been completed.

closed 2023-01-18 by jakemclaughlin6