locusrobotics/fuse #275: ROS2 change of param types, backporting the use of dynamic reconfigure

BrettRD, 2022-06-22:
I'm documenting the parameters used by the various aspects of Fuse, some of them can't be migrated to ros2 because ROS2 has stricter type checking

Specifically, `optimiser/motion_models` and `optimiser/sensor_models` use `XmlRpc::XmlRpcValue::TypeArray` which doesn't exist in ROS2.
In the ROS2 branch, I need to change the yaml structure to declare the names of the motion_models and sensor_models first, then fill in the properties of each afterwards.

examples of a proposed ROS2 compatible format: [fuse_simple_tutorial](https://github.com/BrettRD/fuse/blob/galactic/fuse_tutorials/config/fuse_simple_tutorial.yaml)  [range_sensor_tutorial](https://github.com/BrettRD/fuse/blob/galactic/fuse_tutorials/config/range_sensor_tutorial.yaml)

ROS2 also encourages the use of a built-in callback that is triggered when parameters change.  In ROS1, this was handled by dynamic_reconfigure.
It seems like dynamic_reconfigure would be useful, should I make the effort to streamline a back-port?

methylDragon, 2023-02-07:
With the completion of https://github.com/locusrobotics/fuse/issues/276 , a way to list parameter names has been devised (and [pending upstream](https://github.com/ros2/rclcpp/pull/2090))

So we can now preserve the structure of the yamls!

That said, dynamic_reconfigure-like behavior would be good to include.

Ryanf55, 2023-04-04:
If you need help with dynamic reconfigure, let me know. For the aerial drones, there are a few places we could make use of dynamic reconfiguration for the constrains in the model.