locusrobotics/fuse #276: DRAFT: Fuse -> ROS 2 Rolling

methylDragon, 2022-10-06:
# Description

This issue tracks the progress of the ROS 2 port for Fuse, targeting the `rolling` release of ROS 2.
Some of the work here will be based off the work mentioned in https://github.com/locusrobotics/fuse/issues/257.

# Approach
PRs will be sent against the `rolling` branch on [fuse](https://github.com/locusrobotics/fuse) and [tf2_2d](https://github.com/locusrobotics/tf2_2d), working up the dependency graph and doubling back as necessary.

[image]

# Progress

- [ ] fuse_msgs
- [ ] fuse_core
- [ ] fuse_loss
- [ ] fuse_variables
- [ ] fuse_graphs
- [ ] fuse_constraints
- [ ] fuse_publishers
- [ ] fuse_viz
- [ ] tf2_2d
- [ ] fuse_models
- [ ] fuse_optimizers
- [ ] fuse_tutorials
- [ ] fuse_doc
- [ ] fuse

methylDragon, 2023-01-27:
Fuse functionality is now ported! :tada: 
@svwilliams 

There's still some linting and docs porting to do

sloretz, 2023-01-31:
@methylDragon Is there anything left to do before closing this ticket?

methylDragon, 2023-02-01:
I think we're done here! Ticket closed!!

closed 2023-02-01 by methylDragon