locusrobotics/fuse #318: Extending the sensor processing to 3D

AugusteBourgois, 2023-02-20:
Hi there,

We am currently trying to use fuse to estimate the state of a 3D robot. Unless we are mistaken, the sensor processing models are not available for 3D robots (i.e. IMU, twist, pose messages will be interpreted as 2D).

Do you plan on committing something soon, or should we develop them and propose a PR ? The plan we have right now is to implement 3D models in addition to the exiting 2D models, so we were wondering if you'd be interested in merging the results in a near future.

FYI, we would be at least two qualified robotics engineer from Forssea Robotics working on the matter. We are working with ROS2 Humble on Ubuntu/debian platforms.

Thank you in advance for your answer,

Auguste Bourgois

iblnkn, 2023-03-06:
Hello Auguste, my team and I spent quite a bit of time on this recently and have models working in 3D, though we haven't had the chance to do much testing yet or clean-up yet. It would be great to have some support in this effort. It is currently running on ROS Noetic, though we want to port it to ROS2 soon and would appreciate help on that front. I have not put in a PR yet, but seeing as others are interested, I could make this a priority.

Ryanf55, 2023-04-06:
How can I support this effort?

nachovizzo, 2024-01-18:
Seems like https://github.com/locusrobotics/fuse/pull/354 might be related